MicrosoftDocs/windows-driver-docs #3694: How to Avoid ELAM Errors Disabling a Critical Driver, Causing Endless Blue Screens and Preventing the Computer from Booting

Bzkirito, 2023-10-27:
Hello, while developing an ELAM driver, I envisioned a scenario. Suppose I issued a wrong policy, causing the ELAM driver to mistakenly intercept the startup of some critical system drivers, such as ntfs.sys (of course, this wouldn't be so outrageous in reality, it's just a hypothesis). This would result in endless blue screens on the user's computer and make it difficult to pinpoint the problem.

To address this, I came up with an automated strategy to avoid blue screens, as follows:
    1.Create a BootFlag key in the registry corresponding to the ELAM driver and set its value to 0.
    2.When the ELAM driver starts, it reads this value. If the value is not 0, the ELAM driver does nothing in DriverEntry and returns an error directly.
    3.If the value is 0, change the value to 1. Call IoRegisterBootDriverCallback normally and disable the corresponding drivers according to the policy.
    4.If the system starts normally, a process in the application layer resets the BootFlag to 0.
    5.If a critical component (such as ntfs.sys) is disabled, the blue screen will restart, and the ELAM will not work when the BootFlag value is read as 1. This prevents the user's computer from falling into an endless blue screen predicament.

However, during development, I found that if a blue screen occurs during the execution of the ELAM driver, all modifications to the registry will not take effect.

So, is there any way for me to mark something during the operation of my ELAM driver, allowing me to determine whether a blue screen occurred during the last startup?

Bzkirito, 2023-11-10:
@tedhudek Would you mind helping me?

tedhudek, 2023-11-21:
Sorry @Bzkirito, so far I have been unable to find a good answer for you.  I'm leaving this issue open in case that changes.

Bzkirito, 2023-11-22:
> Sorry @Bzkirito, so far I have been unable to find a good answer for you. I'm leaving this issue open in case that changes.

I've found that during system startup(My be is NT loader), after two consecutive blue screens, Windows will enter an automatic repair state (although it can't successfully repair). So, how does Windows detect during the system boot phase that there were two previous blue screens causing the system to fail to start successfully? If there is a way to detect previous blue screens that caused the system to fail to start during the boot phase, then the problem would be solved.

tedhudek, 2024-01-03:
Apologies, Lloyd, but I don't have anything else for you on this one.  

You may have success posting your question in the following forums, where other driver developers and Microsoft engineers can be found. The community there is quite responsive. Best of luck with this issue.

* [Microsoft Q&A](https://learn.microsoft.com/en-us/answers/questions)
* [OSR's NTDEV forum](https://community.osr.com/categories/ntdev)

closed 2024-01-03 by tedhudek