MicrosoftDocs/windows-driver-docs #3699: If not set Registry Entry "IoTimeoutValue", which timeout value it will use?

wangyan0507, 2023-11-01:
> Name: IoTimeoutValue
> Type: REG_DWORD
> Path:
> Miniport scope: HKLM\System\CurrentControlSet\Services<miniport name>\Parameters
> Value:
> Minimum: 0
> Maximum: 65535
> Units: seconds
> Description: Indicates the I/O timeout value for devices managed by the miniport driver. If this registry value doesn’t exist, the system will use the global disk I/O timeout value.
> Applies: Starting with Windows 8.

In the Description, what does the "the global disk I/O timeout value" mean? It is another Registry Entry?
---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 17df8068-a0e4-4029-eb94-cfd1fa25e59c
* Version Independent ID: 1d664934-d247-f70c-e8e5-86a87a955f85
* Content: [Registry Entries for StorPort Miniport Drivers - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/storage/registry-entries-for-storport-miniport-drivers)
* Content Source: [windows-driver-docs-pr/storage/registry-entries-for-storport-miniport-drivers.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/storage/registry-entries-for-storport-miniport-drivers.md)
* Product: **windows-hardware**
* Technology: **storage**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

lorihollasch, 2023-11-13:
Thanks for the issue, @wangyan0507. I've added clarification for IoTimeoutValue.

closed 2023-11-13 by lorihollasch

wangyan0507, 2023-11-14:
@lorihollasch 

The global disk I/O timeout value is  TimeoutValue or IoTimeoutValue in "HKLM\System\CurrentControlSet\Services\disk\" ?
I don't knows which one is right.

Thanks,
Yan Wang