MicrosoftDocs/windows-driver-docs #3709: Wrong link

RAJU2529, 2023-11-08:
When I click the **debugger command line** link, it goes to **Install the Windows debugger** page

I think the link must be  replaced by this link  **command-line-options.md**
---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 4ffab098-c195-f350-590e-008674d805b4
* Version Independent ID: 2c7ca9d6-2deb-0029-c001-35e921e73fe8
* Content: [Debugging a User-Mode Process Using WinDbg - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/debugging-a-user-mode-process-using-windbg)
* Content Source: [windows-driver-docs-pr/debugger/debugging-a-user-mode-process-using-windbg.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/debugging-a-user-mode-process-using-windbg.md)
* Product: **windows-hardware**
* Technology: **debugger**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

RAJU2529, 2023-11-08:
@DOMARS , What is your opinion on this issue, sir?

DOMARS, 2023-11-10:
@RAJU2529 - This is a valid issue. Somehow the link text had 'the-' at the start, so the link was failing, and it goes to the top level debugger page, which is the install WinDbg page that you mentioned. This has been fixed, and thanks for filing this issue.

closed 2023-11-10 by DOMARS