MicrosoftDocs/windows-driver-docs #3710: HKLM should be replaced

RAJU2529, 2023-11-08:
In this article, i found 7 places **HKLM** is mentioned and  4 places **HKEY_LOCAL_MACHINE** is mentioned 

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: a6ef91fe-7578-43a3-0225-dfa3e753cb23
* Version Independent ID: 95ea21d3-a014-aa1e-5d97-2f77b3702010
* Content: [GFlags Details - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/gflags-details)
* Content Source: [windows-driver-docs-pr/debugger/gflags-details.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/gflags-details.md)
* Product: **windows-hardware**
* Technology: **debugger**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

RAJU2529, 2023-11-08:
@DOMARS, can I replace those **HKLM** to  **HKEY_LOCAL_MACHINE**

DOMARS, 2023-11-10:
@RAJU2529 - Yes, I think it would better to not use the HKLM abbreviation in this table, but rather use the full proper registry entry name - HKEY_LOCAL_MACHINE.

RAJU2529, 2023-11-10:
@DOMARS . I will create PR today.

DOMARS, 2023-11-10:
PR has merged, so closing this issue.

closed 2023-11-10 by DOMARS