MicrosoftDocs/windows-driver-docs #3715: Bluetooth stuck with debug SMP keys after using btvs with Wireshark

grifcj, 2023-11-13:
I downloaded v1.14.0 of the bluetooth test and captured some logs with Wireshark. I ran the command `btvs.exe -Mode Wireshark`, which worked as expected for logging. As part of that activity, I clicked both the Full Packet Logging and the Set or Extend Debug Mode buttons. I believe my issue is associated with the debug mode.

Now, Windows bluetooth stack seems to be stuck using debug SMP keys (assume this was enabled to allow decrypting traffic for packet capture). I can't seem to disable this debug mode, and it's not clear where the setting is even persisted. The button mentions its supposed to be temporary, but doesn't seem to reset after reboots. 

I've tried

- clearing bonds and repairing
- enabling/disabling bluetooth via Windows settings
- enabling/disabling bluetooth adapter via device manager
- rebooting computer
- run uninstaller for bluetooth test platform
- shutdown and restart computer
- system restore computer prior to installing bluetooth test platform (which completed but said it failed to completely restore).

Any guidance here? How do I revert using debug SMP keys?

I'm believe Windows is using debug keys because my peripheral now prints a new log when pairing with Windows, which it didn't before.   

[image]

Pairing with an Android phone continues to work as it did before.

grifcj, 2023-11-13:
I discovered relevant keys in registry under `Computer\HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Services\BTHPORT\Parameters`

Setting `SmpDebugEnabled` and `SimplePairingDebugEnabled` to zero as admin, followed by a reboot, seemed to resolve the issue.

grifcj, 2023-11-18:
@mhopkins-msft 

Missed that tidbit. I have fired off an email, and will get back with you on results, or they may directly comment here.

ArturA-MS, 2023-11-30:
Thanks for bringing this to our attention. The feature team will consider adding functionality to the BTVS tool to allow users to undo the SMP/SimplePairing debug setting as well as expose the current debug setting in the tool. We would like to avoid officially documenting the registry keys for this because those are not officially supported and are subject to change without warning. You should still be able to use them as a workaround in the interim though. I'll also add a warning in the documentation about usage of this BTVS functionality.

grifcj, 2023-11-30:
Sounds good to me. I consider issue resolved on my end for now if want to close ticket.

ArturA-MS, 2023-11-30:
Thank you. Closing.

closed 2023-11-30 by ArturA-MS