MicrosoftDocs/windows-driver-docs #3735: PDB file of ntoskrnl.exe etc missing

BlackINT3, 2023-11-26:
SYMSRV:  HTTPGET: /download/symbols/ntkrnlmp.pdb/8994979D169BB90B3D71598837151C2D1/ntkrnlmp.pdb
SYMSRV:  HttpQueryInfo: 80190194 - HTTP_STATUS_NOT_FOUND

SYMSRV:  HTTPGET: /download/symbols/win32kbase.pdb/A032C5B8405879D387084B592BE7BD301/win32kbase.pdb
SYMSRV:  HttpQueryInfo: 80190194 - HTTP_STATUS_NOT_FOUND

SYMSRV:  HTTPGET: /download/symbols/win32kfull.pdb/ADBDD53613267669E8547855260900CD1/win32kfull.pdb
SYMSRV:  HttpQueryInfo: 80190194 - HTTP_STATUS_NOT_FOUND

OS: Win10 22H2
File Version 10.0.19041.3754
Name ntkrnlmp.exe、win32kfull.sys、win32kbase etc.

Related doc page -- https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/verifying-symbols

The problem similar to #3442.

DOMARS, 2023-12-11:
@BlackINT3 - Assuming the symbol id is for a released version of Windows, I would send an email to [windbgfb@microsoft.com](mailto:windbgfb@microsoft.com) as described on: https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/microsoft-public-symbols

This version of ntkrnlmp is available: https://msdl.microsoft.com/download/symbols/ntkrnlmp.pdb/E9BEBA2FE31582704229E51F816084331/ntkrnlmp.pdb

But this version does not seem to be available currently on the symbol server.
https://msdl.microsoft.com/download/symbols/ntkrnlmp.pdb/8994979D169BB90B3D71598837151C2D1/ntkrnlmp.pdb

We are focused on the docs with these issues, another option for kernel debugging, is stack overflow, or the OSR debugger forum.  For example, this issue was related to network setting. https://community.osr.com/discussion/292092/latest-symbols-of-ntkrnlmp-exe-cannot-be-found-from-microsoft-public-symbol-server

In that thread they used vertarget and lmvm nt, and like you did enabled !sym noisy to gather appropriate information. Good luck with tracking down the symbols you are after.

closed 2023-12-11 by DOMARS

BlackINT3, 2024-01-15:
@DOMARS I appreciate that, but ntkrnlmp.exe still isn't working. I uploaded the file. Could you please help verify it? Thanks!

[ntoskrnl.zip](https://github.com/MicrosoftDocs/windows-driver-docs/files/13935592/ntoskrnl.zip)

DOMARS, 2024-01-16:
Thanks for reaching out to the Windows Driver Developer Documentation team. Unfortunately, we don't have the resources to offer individual technical advice to the broad array of customers who use the driver documentation.

You may have success posting your question in the following forums, where other driver developers and Microsoft engineers can be found. 

[Microsoft Support Community](https://answers.microsoft.com/)

[OSR's windbg List](https://community.osr.com/categories/windbg)

This feedback mechanism is for requests about the documentation. Best of luck with this issue.