MicrosoftDocs/windows-driver-docs #3736: Need more details on Format NVM translations

vonericsen, 2023-11-27:
Hello,

Can you provide more information on how to use the translations for SCSI Sanitize and SCSI Format unit to the format NVM command?
I have tried using the translation according to the whitepaper from nvmexpress.org to issue a NVM format command, but have not been able to get it to work.
The sense data returned by the driver for the Format Unit CDB is "Illegal Request - Invalid Command Operation Code", which tells me this translation is not actually implemented. I've tried this on Windows 10 and Windows 11 with the latest updates.

For the Sanitize CDB, this is not described in the translation whitepaper, so I assume this is a Microsoft unique translation to the format command and I would like more details on how to use this command and which service actions are allowed and what they translate to in a Format NVM command.

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: f2d95d67-adbc-edf7-dd5f-4f691f4268ab
* Version Independent ID: c280bf38-106d-041c-342e-a8aea2e7a645
* Content: [StorNVMe SCSI Translation Support - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/storage/stornvme-scsi-translation-support)
* Content Source: [windows-driver-docs-pr/storage/stornvme-scsi-translation-support.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/storage/stornvme-scsi-translation-support.md)
* Product: **windows-hardware**
* Technology: **storage**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

vonericsen, 2024-01-04:
I have had some time to do more testing and I am a bit puzzled about this translation overall based on what I have found with my testing in Windows 11.

I have a drive that supports sanitize crypto erase and sanitize block erase but does not support the NVM format with crypto erase.
If I issue the CDB for SCSI Sanitize Block erase, it translates to NVMe Sanitize block erase.
If I issue the CDB for SCSI Sanitize Crypto erase, it comes back with an error. I assume that this is returning an error because instead of translating to NVMe sanitize crypto erase, it attempts to translate to NVMe Format with crypto erase instead....but because this device does not support format with crypto erase, it gets an error.

Because of this observed behavior, it seems that the sanitize CDB can translate to either format or sanitize but this could depend on the device's capabilities or something else.

There is some work by the [T10](t10.org) committee to begin standardization of SCSI to NVMe translation, so it would be important to clarify the current translations for Sanitize (and whenever they change) as I expect the committee will specify that a SCSI sanitize command only translates to an NVM sanitize command.
Broadcom has shared a document of their translations to the T10 commitee showing SCSI Sanitize only translates to NVMe Sanitize.

It is important to understand which command is being translated to because of the IEEE 2883 standard for data sanitization.
Sanitize commands meet "purge", however NVMe Format with crypto erase does NOT meet "purge" per this specification and only qualifies as a "clear". 
For those coming from a SCSI command set background, the expectation would be that any SCSI Sanitize command meets purge, so translating to a non-purge command should be clearly documented.

I hope this feedback can help with this issue and figuring out how to update the documentation.

lorihollasch, 2024-01-12:
@vonericsen, apologies for my slow response. I was finally able to get some additional information about SCSI->NVMe Sanitize and updated the page accordingly. In coming to close this issue, I'm just now seeing your comment from last week. I hope the additional info on the page helps. If anything more is missing, feel free to open a new issue.

closed 2024-01-12 by lorihollasch