MicrosoftDocs/windows-driver-docs #3737: Some clarification of the role of FltVetoBypassIo might help

rodwiddowson, 2023-11-27:
Lori, Ted,

It might be nice if you mentioned that the whole reason that FltVetoBypassIo exists is to setup the `FS_BPIO_OUTPUT` buffer for your (or such is my observation by nature study).

If this is wrong then  it would be helpful to be more concrete about what the call does do and what is left for you to do.

Thanks, take care
---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 9dd321d8-dd23-8f32-3763-9b43fa484259
* Version Independent ID: 9dd321d8-dd23-8f32-3763-9b43fa484259
* Content: [Supporting BypassIO operations - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/ifs/bypassio-operations)
* Content Source: [windows-driver-docs-pr/ifs/bypassio-operations.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/ifs/bypassio-operations.md)
* Product: **windows-hardware**
* Technology: **ifs**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

lorihollasch, 2023-12-05:
@rodwiddowson, great call out, and on further investigation it turns out that the docs didn't match the code. I've updated this page accordingly for FltVetoBypassIo's role during ENABLE (and QUERY). Thanks for reporting this!

lorihollasch, 2023-12-05:
@rodwiddowson, I forgot to mention that I also updated [FltVetoBypassIo](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/fltkernel/nf-fltkernel-fltvetobypassio) accordingly.

closed 2023-12-05 by lorihollasch