MicrosoftDocs/windows-driver-docs #3740: Are ECDSA/ECC certificates supported on hlkx file signatures?

olljanat, 2023-11-29:
Background, more than two years ago SSL.com (which is one of the supported EV code certificate providers) [switched to ECDSA/ECC certificates](https://www.ssl.com/blogs/new-minimum-rsa-key-size-for-code-signing-certificates/) instead of RSA.

Based on code examples I managed to create [tool](https://github.com/dotnet/runtime/issues/36103#issuecomment-1828510749) which is able to sign hlkx files with certificate they provided in Yubikey FIPS.

With that my submission (Submission ID: 1152921505697077071 ) did pass "Package Acceptance" step but failed on "Preparation" step
[image]
and only thing which exists error log is message `Signature validation failed while opening the pacakge`

Question is that should this be supported?

Also it looks to be that my submission is now stuck in way that I cannot update it but it also didn't went to manual review which looks like bug on portal.

PS. Example documentation page about this topic is https://learn.microsoft.com/en-us/windows-hardware/test/hlk/user/hlk-signing-with-an-hsm#packagedigitalsignaturemanager but it looks to be missing submit feedback button.

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 855a80c4-930e-4609-f05d-c516e9057135
* Version Independent ID: 855a80c4-930e-4609-f05d-c516e9057135
* Content: [Update a hardware submission - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/hardware-submission-update)
* Content Source: [windows-driver-docs-pr/dashboard/hardware-submission-update.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/hardware-submission-update.md)
* Product: **windows-hardware**
* Technology: **dashboard**
* GitHub Login: @anaharris-ms
* Microsoft Alias: **anaharris**

anaharris-ms, 2023-11-30:
Hi @olljanat,

So sorry you are having trouble. As this is not a doc issue, you need to contact customer support here: [Technical support issues](https://learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/hardware-submission-support#technical-support-issues).

closed 2023-11-30 by anaharris-ms

olljanat, 2023-11-30:
@anaharris-ms it is doc issue on that sense that there is no any mentioning that which certificate types are/aren't supported. Page https://learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/code-signing-reqs#where-to-get-code-signing-certificates clearly says that SSL.com certificates are supported and their page and their page https://www.ssl.com/certificates/ev-code-signing/buy/ says that those can be used signing kernel drivers.

I also created ticket for them and got answer:
> hlkx file can only be signed and submitted with an RSA issued EV Code Signing Certificate. I believe you signed your using Yubikey. Yubikeys only supports ECDSA/ECC algorithms. RSA Certificates can only be issued through an HSM or eSigner.

and if that really is hard requirement it should be written to documentation.

SSL.com EV code signing certificate was cheapest option when ordered with Yubikey but their competitors have better price on HSM/Cloud Signing products.

Sure I can contact technical support too but still this should be publicly documented.