MicrosoftDocs/windows-driver-docs #3741: Override of monitor EDIDs separately based on HDMI or DisplayPort interface?

powerlands, 2023-11-30:
Hi,

The page of "Manufacturer override of monitor EDIDs" is very helpful to understand how to do override. Thanks!

However, in general, EDIDs of HDMI and DisplayPort from one monitor model are different. It looks like that the current overriding mechanism overrides EDID no matter if it's from HDMI or DP to the same values. Is there a way to override based on the interface?

Thanks,
Yan

zhuman, 2023-11-30:
Hi @powerlands, that's a great question. We don't support separately overriding the same monitor on HDMI vs DisplayPort today unfortunately, at least not as a feature of the OS. Some monitors may expose different device IDs over HDMI and DisplayPort separately, so that would work since they would get separate device instances, but if they report the same device ID then we use the same device instance so the same override registry key would apply either way. And reporting the same device ID from each connector type is actually what we prefer since the user typically doesn't care about the connector type, they want their settings to be persisted either way.

It's not a bad idea to consider for the future though, I can understand why there might be cases you'd want to override the HDMI block specifically.

powerlands, 2023-12-01:
Thank you, Zach!

lorihollasch, 2023-12-14:
Thanks for the issue, @powerlands, and thanks for the response, @zhuman! Closing this now.

closed 2023-12-14 by lorihollasch