MicrosoftDocs/windows-driver-docs #3745: The Security Checklist renders empty check boxes, with no topics

viorel-m, 2023-12-09:
### Type of issue

Typo

### Feedback

[image]

### Page URL

https://learn.microsoft.com/en-us/windows-hardware/drivers/driversecurity/driver-security-checklist

### Content source URL

https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/driversecurity/driver-security-checklist.md

### Author

@DOMARS

DOMARS, 2023-12-09:
@viorel-m - Thanks for filing this issue. I see the same content drop out and will work to restore the checklist items. In the mean time, the checklist items, correspond to the section headings at the top of the topic: 

[Confirm that a kernel driver is required](https://learn.microsoft.com/en-us/windows-hardware/drivers/driversecurity/driver-security-checklist#confirm-that-a-kernel-driver-is-required)
[Use the driver frameworks](https://learn.microsoft.com/en-us/windows-hardware/drivers/driversecurity/driver-security-checklist#use-the-driver-frameworks)
[Control access to software only drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/driversecurity/driver-security-checklist#control-access-to-software-only-drivers)
[Do not production sign test code](https://learn.microsoft.com/en-us/windows-hardware/drivers/driversecurity/driver-security-checklist#do-not-production-sign-test-code)
[Perform threat analysis](https://learn.microsoft.com/en-us/windows-hardware/drivers/driversecurity/driver-security-checklist#perform-threat-analysis)
[Follow driver secure coding guidelines](https://learn.microsoft.com/en-us/windows-hardware/drivers/driversecurity/driver-security-checklist#follow-driver-secure-coding-guidelines)
[Manage driver access control](https://learn.microsoft.com/en-us/windows-hardware/drivers/driversecurity/driver-security-checklist#manage-driver-access-control)
[Security Identifiers (SIDs) risk hierarchy](https://learn.microsoft.com/en-us/windows-hardware/drivers/driversecurity/driver-security-checklist#security-identifiers-sids-risk-hierarchy)
[Implement HVCI compatible code](https://learn.microsoft.com/en-us/windows-hardware/drivers/driversecurity/driver-security-checklist#implement-hvci-compatible-code)
[Follow technology-specific code best practices](https://learn.microsoft.com/en-us/windows-hardware/drivers/driversecurity/driver-security-checklist#follow-technology-specific-code-best-practices)
[Enhance device installation security](https://learn.microsoft.com/en-us/windows-hardware/drivers/driversecurity/driver-security-checklist#enhance-device-installation-security)
[Perform peer code review](https://learn.microsoft.com/en-us/windows-hardware/drivers/driversecurity/driver-security-checklist#perform-peer-code-review)
[Execute proper release driver signing](https://learn.microsoft.com/en-us/windows-hardware/drivers/driversecurity/driver-security-checklist#execute-proper-release-driver-signing)
[Use code analysis in Visual Studio to investigate driver security](https://learn.microsoft.com/en-us/windows-hardware/drivers/driversecurity/driver-security-checklist#use-code-analysis-in-visual-studio-to-investigate-driver-security)
[Use Static Driver Verifier to check for vulnerabilities](https://learn.microsoft.com/en-us/windows-hardware/drivers/driversecurity/driver-security-checklist#use-static-driver-verifier-to-check-for-vulnerabilities)
[Check code with the BinSkim Binary Analyzer](https://learn.microsoft.com/en-us/windows-hardware/drivers/driversecurity/driver-security-checklist#check-code-with-the-binskim-binary-analyzer)
[Use additional code validation tools](https://learn.microsoft.com/en-us/windows-hardware/drivers/driversecurity/driver-security-checklist#use-additional-code-validation-tools)
[Review debugger techniques and extensions](https://learn.microsoft.com/en-us/windows-hardware/drivers/driversecurity/driver-security-checklist#review-debugger-techniques-and-extensions)
[Microsoft Vulnerable and Malicious Driver Reporting Center](https://learn.microsoft.com/en-us/windows-hardware/drivers/driversecurity/driver-security-checklist#microsoft-vulnerable-and-malicious-driver-reporting-center)
[Review secure coding resources](https://learn.microsoft.com/en-us/windows-hardware/drivers/driversecurity/driver-security-checklist#review-secure-coding-resources)

viorel-m, 2023-12-09:
Thank you, Don - I appreciate it!

DOMARS, 2023-12-12:
@viorel-m - The in-topic links that are the driver security checklist, have been restored. Thanks again for filing this issue and helping to make the Windows driver docs better.

closed 2023-12-12 by DOMARS