MicrosoftDocs/windows-driver-docs #3753: Customer feedback - COSA documentation is incomplete

samyun, 2023-12-21:
### Type of issue

Missing information

### Feedback

The COSA documentation is missing information / incorrect in some places. For example, the `HotSpot` field is not correct. 

### Page URL

https://learn.microsoft.com/en-us/windows-hardware/drivers/mobilebroadband/desktop-cosa-apn-database-settings

### Content source URL

https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/mobilebroadband/desktop-cosa-apn-database-settings.md

### Author

@aviviano

aviviano, 2023-12-22:
Hi @mhopkins-msft - Is this topic a part of the COSA updates you're working on?

aviviano, 2024-01-02:
> > Hi @mhopkins-msft - Is this topic a part of the COSA updates you're working on?
> 
> Yes, updates are underway.

Thanks - I assigned this issue to you.