MicrosoftDocs/windows-driver-docs #3769: Regarding the topic "Promoting a Driver's StartType at Boot Depending on Boot Scenario" for USB4

GeorgeGTA, 2024-01-12:
### Type of issue

Other (describe below)

### Feedback

In this topic [Specifying Driver Load Order | Microsoft Docs](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/specifying-driver-load-order#using-bootflags-to-promote-a-drivers-starttype-at-boot-depending-on-boot-scenario) , it mentions we can use BootFlags to Promote a Driver's StartType at Boot Depending on Boot Scenario, but how about USB4 if a driver should be promoted while booting from a disk on USB4 controller ? what a hexadecimal value should be specified accordingly? 
Right now I am trying to boot Windows 11(23H2) installed on a SSD through USB4 port but get BSOD "Inaccessible Boot Device“. I can boot the same disk on USB3.2 port with the same laptop Legion 5i Pro, but cannot boot on USB 4 port, although it can be identified by bios and can be selected from the bios boot option list.

### Page URL

https://learn.microsoft.com/en-us/windows-hardware/drivers/install/specifying-driver-load-order#using-bootflags-to-promote-a-drivers-starttype-at-boot-depending-on-boot-scenario

### Content source URL

https://learn.microsoft.com/en-us/windows-hardware/drivers/install/specifying-driver-load-order#using-bootflags-to-promote-a-drivers-starttype-at-boot-depending-on-boot-scenario

### Author

George

GeorgeGTA, 2024-01-16:
Attache two screenshots for the information about the drivers installed which were caught after booting on USB3.2.
[image]
[image]

tedhudek, 2024-01-24:
Hi @GeorgeGTA thanks for the question.  Performing a Windows OS installation using a Thunderbolt/USB4 storage device or a USB3 drive behind USB4 dock is supported and should work fine.
                                                                                               
However, booting Windows from a portable Thunderbolt storage device is not supported.

Hope this helps and best of luck!

closed 2024-01-24 by tedhudek