MicrosoftDocs/windows-driver-docs #3783: Customer feedback - Step 1 of winget WDK installation requires absolute path

wmmc88, 2024-01-26:
### Type of issue

Missing information

### Feedback

The VS installer requires an absolute path to `wdk.vsconfig`. Providing a valid relative path will cause winget to install the Visual Studio Installer successfully, but it will silently fail the Visual Studio installation. Winget reports the installation as a success.

See VS installer bug: https://developercommunity.visualstudio.com/t/When-installing-Visual-via-winget-and-a-/10572529

I think the docs should be updated to emphasize that `<vsconfig-folder>` listed in the examples **must** be an absolute path

### Page URL

https://learn.microsoft.com/en-us/windows-hardware/drivers/install-the-wdk-using-winget#step-1-install-visual-studio-2022

### Content source URL

https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install-the-wdk-using-winget.md

### Author

@tedhudek

tedhudek, 2024-01-29:
Hi @wmmc88 thanks for the feedback.  Would you be interested in opening a PR with your proposed feedback?  To get started select the Edit icon in the upper right of the doc page in question, and it will take you to the relevant .md file in GitHub, from which select Edit again and away you go...

tedhudek, 2024-02-02:
Hi Melvin, I went ahead and updated https://learn.microsoft.com/en-us/windows-hardware/drivers/install-the-wdk-using-winget#step-1-install-visual-studio-2022 to fix this.  See the red Caution in there now.  Thanks again for the report!

closed 2024-02-02 by tedhudek