MicrosoftDocs/windows-driver-docs #3784: Customer feedback - winget Windows SDK install command fails due to long log path

wmmc88, 2024-01-26:
### Type of issue

Missing information

### Feedback

The current SDK install command listed in step 2 will fail because of log paths exceeding the path limit. See https://github.com/microsoft/winget-cli/issues/2143#issuecomment-1912747651. 

I ran into this issue on all the devices I tested it on, so it may be good to call out the `--log` workaround from the linked comment in the official winget WDK install docs.

CC @NeoAdonis 

### Page URL

https://learn.microsoft.com/en-us/windows-hardware/drivers/install-the-wdk-using-winget

### Content source URL

https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install-the-wdk-using-winget.md

### Author

@tedhudek

tedhudek, 2024-01-29:
Same comment as #3783

tedhudek, 2024-02-02:
Thanks also for this one Melvin.  I added the --log parameter to both the SDK and WDK install command lines in https://learn.microsoft.com/en-us/windows-hardware/drivers/install-the-wdk-using-winget#step-2-install-windows-sdk-and-wdk.  Thanks for caring about the docs!

closed 2024-02-02 by tedhudek