MicrosoftDocs/windows-driver-docs #3793: One section of original midriver v6.02 changes list is missing in the online document

CryptoTechSA, 2024-02-02:
### Type of issue

Missing information

### Feedback

Comparing .DOC complete specification of smart card miniDriver v6.02 with this online content (https://learn.microsoft.com/en-us/windows-hardware/drivers/smartcard/minidriver-version-6-02-features), I have noticed missed part:
"1.1.2	Support for Read-Only Cards
Smart cards are considered read-only when Windows cannot write specific cache data to the card. The definition of a read-only card is included in section 7.4, “Read-Only Cards.”
Version 6 adds new modes of data caching and enables a card minidriver to control those cache modes.  For more details, see section 4.5.4, “API CardGetProperty.”
With these new data cache modes, along with proper virtualization of other system files (see section 5.2, “File System Virtualization”), it is now possible to develop a card minidriver that is fully compatible with a read-only smart card.
For more information, see section 7.4, “Read-Only Cards,” and section 4.5.4, “CardGetProperty.”
1.1.3	Secure PIN Channel
Secure PIN channel is a feature in Windows Vista SP1 that enables ......."

### Page URL

https://learn.microsoft.com/en-us/windows-hardware/drivers/smartcard/minidriver-version-6-02-features

### Content source URL

https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/smartcard/minidriver-version-6-02-features.md

### Author

@tedhudek

tedhudek, 2024-02-02:
Thank you for letting us know, @CryptoTechSA!  Would you happen to have a link to the spec you are looking at?  It would be helpful to reference it as we update the online document.

CryptoTechSA, 2024-02-05:
Sure. It is not easy to find (as many valuable smart card related content which is still indexed at search engines but Microsoft's sites links do not work any more and new linking is not present :(  
I have eventually found it here -> https://download.microsoft.com/download/9/c/5/9c5b2167-8017-4bae-9fde-d599bac8184a/sc-minidriver_specs_V6-final.docx