MicrosoftDocs/windows-driver-docs #3807: Clarification needed

rodwiddowson, 2024-02-15:
### Type of issue

Missing information

### Feedback

Firstly.  The github page to report errors is still not filling in the URLs and owner

Now the questions

1.  You say " A filter must register for this operation." ? really?  That is true of old style filters but surely not of minifilters.  And if it is true it should say "Minifilters must register....
2. It looks like it is possible to reject an MJ_QUERY_OPEN in the post call with `FLT_POSTOP_DISALLOW_FSFILTER_IO` (although that is not documented and maybe should).  Can you return this status for MJ_NETWORK_QUERY (I'm about to find out)

### Page URL

https://learn.microsoft.com/en-us/windows-hardware/drivers/ifs/flt-parameters-for-irp-mj-network-query-open

### Content source URL

https://learn.microsoft.com/en-us/windows-hardware/drivers/ifs/flt-parameters-for-irp-mj-network-query-open

### Author

@lorihollasch

rodwiddowson, 2024-02-15:
Update.  I see that `FLT_POSTOP_DISALLOW_FSFILTER_IO` is documented, just not on this page..

lorihollasch, 2024-02-26:
@rodwiddowson:
1. Deleted that "must register" sentence.
2. I added the DISALLOW conditions to OPEN and NETWORK_OPEN pages.
Regarding GitHub issue auto-fill: I'm guessing nothing will be done about that since issues will be turned off in a few weeks (PRs will still be accepted).
Thanks for reporting these!

closed 2024-02-26 by lorihollasch