MicrosoftDocs/windows-driver-docs #3812: Feedback

Beanv587, 2024-02-28:
### Type of issue

Other (describe below)

### Feedback

Hello, I want to register for the Hardware Developer Program. Follow the steps on this page (https://learn.microsoft.com/zh-cn/windows-hardware/drivers/dashboard/get-started-dashboard-submissions?source=recommendations #step-2-register-for-the-hardware-developer-program),Open this link https://go.microsoft.com/fwlink/?LinkID=828002. The redirected page is not the page to register for the Hardware Developer Program...

In fact, 1 month ago this page should have redirected to https://partner.microsoft.com/en-us/dashboard/Registration/Tenant/CreateTenant

But I don’t know why even if I manually open this page https://partner.microsoft.com/en-us/dashboard/Registration/Tenant/CreateTenant, I cannot complete the registration. After I fill in the information, the page will exit and let me Login again.. I cannot proceed to the Certificate step...

I don't know what happened, but I hope it can be fixed soon.Thank you

### Page URL

https://go.microsoft.com/fwlink/?LinkID=828002

### Content source URL

https://learn.microsoft.com/zh-cn/windows-hardware/drivers/dashboard/get-started-dashboard-submissions?source=recommendations#step-2-register-for-the-hardware-developer-program

### Author

anaharris-ms

jianghui3132749, 2024-03-04:
i have the same question. i want to regiest Hardware Developer, but i can not.

anaharris-ms, 2024-03-05:
Hi @Beanv587, thank you so much bringing this to our attention. I've fixed the link, and things should be working now. Let me know if you need anything else.

#please-close

Beanv587, 2024-03-05:
> Hi @Beanv587, thank you so much bringing this to our attention. I've fixed the link, and things should be working now. Let me know if you need anything else.
> 
> #please-close

I still can't register for the Hardware Developer Program. When I visit https://go.microsoft.com/fwlink/?LinkID=828002, it prompts me to register a Microsoft account. After I register the account, it asks me to join a program. But there is no Hardware Developer Program I want...

[image]

[image]

Beanv587, 2024-03-05:
If I sigin using my work account, the page looks like this:

[image]

Still no hardware developer program

Beanv587, 2024-03-05:
@anaharris-ms

jianghui3132749, 2024-03-05:
yes，i have the same question.  Still no hardware developer program

Beanv587, 2024-03-05:
> yes，i have the same question. Still no hardware developer program

Hope they fix it soon

anaharris-ms, 2024-03-05:
Hi @jianghui3132749 and @Beanv587, I am looking into this for you. It's late here, but I will be hopefully resolving this issue for your tomorrow! I deeply apologize for any inconvenience.

Beanv587, 2024-03-05:
> Hi @jianghui3132749 and @Beanv587, I am looking into this for you. It's late here, but I will be hopefully resolving this issue for your tomorrow! I deeply apologize for any inconvenience.

Thank you

anaharris-ms, 2024-03-05:
Hi, I have spoken to the internal team here. For this issue, you'll need to contact support at https://review.learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/hardware-submission-support?branch=pr-en-us-2301#technical-support-issues.

I am so sorry for any inconvenience.

jjwwiodyne, 2024-03-05:
@anaharris-ms When I try to click on Developer Support I get "The request is blocked.
0Km/nZQAAAAAZc/72KMslTZyJLhSF99wdQ082QUEzMTUwODA3MDE3ADg1MjU3MDQwLWY4YzUtNDFkZS05MTdiLWM1NzJiZGM5MDBiYg=="

anaharris-ms, 2024-03-05:
@jjwwiodyne, 

Try this: https://learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/hardware-submission-support?branch=main

jjwwiodyne, 2024-03-05:
Thanks @anaharris-ms for your assistance. I'm still seeing it
blocked...here's a screencast: https://share.zight.com/NQuBpQl9

Is there a support email or phone number we can call? This is getting
really frustrating sorting this out on our own.

anaharris-ms, 2024-03-05:
@jjwwiodyne,

I will look into this as well. Hope to get back to you later this evening. Thanks for your patience!

jjwwiodyne, 2024-03-05:
@anaharris-ms Thank you! We're eager to get our drivers signed.

anaharris-ms, 2024-03-05:
@jjwwiodyne, Can you check to see if there are any ad blockers or something else on your end. I am having no problems accessing the site, and support is not having any access issues.

jjwwiodyne, 2024-03-05:
@anaharris-ms It appears that the site only works with Microsoft Edge not
Chrome or Safari on Mac. I was able to submit a ticket now. Thank you.

anaharris-ms, 2024-03-05:
@jjwwiodyne,

I'm glad you got that resolved. I will let them know that it doesn't work on Chrome or Safari on Mac.

closed 2024-03-05 by anaharris-ms

jianghui3132749, 2024-03-06:
@jjwwiodyne  The response from the support team to me

Hi Team,
 
We are aware that the hardware registration page is currently unavailable. We apologize for any inconvenience this may have caused. Our team is actively working to remediate the issue and restore the page as soon as possible. However, at this time we do not have an estimated time of resolution.
 
I will proceed to close the ticket.
 
Kindest Regards,
Juan
Your Partners at Microsoft
Office Hours 8AM – 5PM US CT
v-juangue@microsoft.com
[Contact Support](https://developer.microsoft.com/en-us/windows/support/?tabs=keytopics)

jianghui3132749, 2024-03-06:
this is a photo

[image]

Beanv587, 2024-03-06:
> Hi, I have spoken to the internal team here. For this issue, you'll need to contact support at https://learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/hardware-submission-support?branch=main
> 
> I am so sorry for any inconvenience.

OK,Thanks

Beanv587, 2024-03-06:
> Hi, I have spoken to the internal team here. For this issue, you'll need to contact support at https://learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/hardware-submission-support?branch=main
> 
> I am so sorry for any inconvenience.

@anaharris-ms OK,Thanks

Beanv587, 2024-03-06:
> jianghui3132749

@jianghui3132749 looks like we have to wait for a while...

jjwwiodyne, 2024-03-14:
@anaharris-ms Any word when we can enroll into the hardware program? The support case I opened 9 days ago just told me to wait. We seriously need to get our drivers signed or this is going to cost us customers Is there a phone number we can escalate this at? Feel free to PM me.

Beanv587, 2024-03-16:
> @anaharris-ms Any word when we can enroll into the hardware program? The support case I opened 9 days ago just told me to wait. We seriously need to get our drivers signed or this is going to cost us customers Is there a phone number we can escalate this at? Feel free to PM me.

I'm in the same situation as you. The team is waiting for my driver signature. . .