MicrosoftDocs/windows-driver-docs #3814: Feedback

jianghui3132749, 2024-03-04:
### Type of issue

Typo

### Feedback

i want to regiest Hardware Developer, i click "Go to the Hardware Developer Program registration page", i can not find prerequisites page. please help me. thanks.

### Page URL

https://learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/hardware-program-register

### Content source URL

https://learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/hardware-program-register

### Author

 jianghui3132749

anaharris-ms, 2024-03-05:
Hi @jianghui3132749, thank you so much bringing this to our attention. I've fixed the link, and things should be working now. Let me know if you need anything else.

#please-close

closed 2024-03-05 by anaharris-ms

jianghui3132749, 2024-03-05:
hi, i want to register windows Hardware Developer, which one is right?  "windows and xbox" or "bethesda partner program"?

[image]
https://partner.microsoft.com/en-us/dashboard/account/exp/enrollment/programselector/AppDev
https://learn.microsoft.com/zh-cn/windows-hardware/drivers/dashboard/hardware-program-register