MicrosoftDocs/windows-driver-docs #3815: Clicking on Hardware Developer Program link just loops.

jjwwiodyne, 2024-03-04:
### Type of issue

Other (describe below)

### Feedback

On this page:
https://learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/hardware-program-register

Clicking on the link (https://go.microsoft.com/fwlink/?LinkID=828002) for the "Hardware Developer Program registration page" just loops back to:
https://learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/

### Page URL

https://learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/hardware-program-register

### Content source URL

https://learn.microsoft.com/en-us/windows-hardware/drivers/dashboard/hardware-program-register

### Author

anaharris-ms

jianghui3132749, 2024-03-05:
you can try it again. i register windows Hardware Developer success.

jianghui3132749, 2024-03-05:
but i have another question.
https://github.ink/MicrosoftDocs/windows-driver-docs/issues/3816#issue-2168121628

jjwwiodyne, 2024-03-05:
Actually it redirects to the "Windows and Xbox program" now instead of Hardware.

jianghui3132749, 2024-03-05:
yes，there is no option for hardware developer. 
hope microsoft technician help us。

anaharris-ms, 2024-03-05:
I am closing this issue, as it is a duplicate of https://github.com/MicrosoftDocs/windows-driver-docs/issues/3812.

I will get back to you on this later in the day.

closed 2024-03-05 by anaharris-ms